ecocurious2/MultiGeiger #443: Help with a project based on this: is my Schematic wrong?

Johannes-ece, 2021-10-02:
[image]
Is this is not the right site for asking the question, I am sorry. I think I don´t get a high signal from the Cap_Full but I cant explain why.
Geiger in is the + pole and Geiger out the -.

rexfue, 2021-10-02:
Hello, your schematic look OK. What types (values) are the diodes D1, D2 and D3 ? They all should be zener diodes, D1 and D2  ZY200 and D3 BZX55 3V3. Hope, that helps.

ThomasWaldmann, 2021-10-02:
Tagged this as question and wontfix, because it is not an issue of the multigeiger.

ThomasWaldmann, 2021-10-02:
C2: 10 or 100nF? Same for C3.

R1: 1W, really? for what?

If you measure resistance over R1, is it really 10kOhm?

Johannes-ece, 2021-10-03:
Ok, so I checked all the parts values again and they math the ones in the partlist from here: https://github.com/ecocurious2/MultiGeiger/tree/master/docs/hardware
to test the circuit I tried to use this Arduino sketch with a different baud rate: https://www.boehri.de/elektronik/Simple_Arduino_Geiger/Simple_Adruino_Geiger.ino
I put some prints in the interrupt to check when they are triggered, and the GMZ_capacitor_full interrupt never triggers, the GMZ_count interrupt subsequently also never triggers.
I know this isn´t exatly this github´s scope, but my circuit also messures co2 and fine particles so I made a custom circuit board.
So thank you so much for your help

WwwBoehriDe, 2021-10-04:
To me everything looks fine. If you have used the components we use, it should work: https://github.com/ecocurious2/MultiGeiger/blob/master/docs/hardware/MultiGeiger_Partlist.ods. The partselection is critical. For example regaring leakagecurrents of the caps and maximum voltage of our coil (happyly it widstands 400 V - even though it is not specified   :-(   ).
Do you have leakage-currents? For example weakly conductive flux from soldering? This could be one problem. If yes: Cleaning with alcohol could help. I normally debug the HV-Part like this: Remove R2 - just to be sure that no current escapes on that path. Write a sketch that generates always pulses on the HF_FET_OUT. Now you should see a voltage on C3. I use a 1 G Ohm Pre-Resistor on my voltmeter. Otherwise the Voltage could drop - our voltage-generation is quite weak. The voltage schould be about 403 V. And you should see the voltage seperation on the Z-Diodes accordingly: 200 V, 200 V, 3.3 V - so the HV_CAP_FULL should deliver a 3.3 V High-Level.
I hope this helps for the first time. Best regards  Juergen

WwwBoehriDe, 2021-10-05:
Hi,
well with a multimeter it is not easy to debug. But anyhow - it already looks like if you are on the right path.

OK, 360 V on C3 looks half-good. Yes, probably it breaks down if you touch it with your multimeter - but we don't know. I would recommend first to try to generate 203 V for a trial. So if you short circuit D1 you should see 203 V on C3.

If this works there should be a short current pulse through D1, D2, D3. And this pulse would generate a short voltage-spike on D3 (= HV_CAP_FULL). But measuring that with a multimeter is not possible. Probably, if you're happy, you would see the average voltage - so maybe 0,1 Volts - but probably you don't see anything. But these spikes should be capable to trigger the interrupt of your Arduino.

I hope this helps. Good luck. Best regards    Juergen

Johannes-ece, 2021-10-17:
I think it works! 
I tried 3 diffrent pcbs and it seems like two of them had a little short somewhere.
[image]
Only the HV Pulses on the far right don't make sense to me, I expected more.

Thank you so much for all your help!