ecocurious2/MultiGeiger #452: frequent reboots on v1.16.0 (Core panic)

jkneer, 2022-03-08:
I encounter frequent reboots of multigeiger on v1.16.0. The attached serial log has 2 reboots within a minute at the end.

```
10:33:17.335 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
10:33:17.335 -> Core 1 register dump:
10:33:17.335 -> PC      : 0x401a712c  PS      : 0x00060c30  A0      : 0x800fa936  A1      : 0x3ffce880  
10:33:17.335 -> A2      : 0x3ffbb9f0  A3      : 0x00000010  A4      : 0x3ffc5da4  A5      : 0xb3069441  
10:33:17.372 -> A6      : 0x7f000000  A7      : 0x7f000000  A8      : 0x7f000000  A9      : 0x3ffce870  
10:33:17.372 -> A10     : 0x3ffdeb78  A11     : 0x00000000  A12     : 0x0000002c  A13     : 0x3ffdcf5c  
10:33:17.372 -> A14     : 0x3ffcec74  A15     : 0x00000002  SAR     : 0x0000000c  EXCCAUSE: 0x0000001c  
10:33:17.372 -> EXCVADDR: 0x00000020  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  
10:33:17.372 -> 
10:33:17.372 -> Backtrace: 0x401a712c:0x3ffce880 0x400fa933:0x3ffce8a0 0x400fa9a5:0x3ffce8c0 0x400fbb89:0x3ffce8e0 0x400d47d9:0x3ffce920 0x400d4c33:0x3ffce960 0x400d4f16:0x3ffcedb0 0x400d4076:0x3ffcee00 0x400d4175:0x3ffcee60 0x400ffcb9:0x3ffceeb0 0x4008e781:0x3ffceed0
10:33:17.408 -> 
10:33:17.408 -> Rebooting...
```

[multigeiger-serial-log-2reboots.txt](https://github.com/ecocurious2/MultiGeiger/files/8204710/multigeiger-serial-log-2reboots.txt)

jkneer, 2022-03-10:
similar or same bug as [issue 314](https://github.com/ecocurious2/MultiGeiger/issues/314)

ThomasWaldmann, 2022-03-10:
suspicion from last video conf: is it madavi related? todo: test with madavi transmission on vs. off.