ecocurious2/MultiGeiger #454: Fehler beim Kompilieren - "WiFi.h"

zwiebelxxl, 2022-04-09:
Hallo MultiGeiger Team,

ich bin total begeistert von dem Projekt und und hab mir Platine/Bauteile + Heltec WiFi Kit 32 besort. 

Zuerst hab ich mit esptool + multigeiger.bin wie in [https://github.com/ecocurious2/MultiGeiger/issues/198](url) beschrieben. Aber leider hängt der esp32 in einem reboot loop.

Also hab ich mich an die Anleitung gehalten und Arduino mit allen Bibliotheken instaliert.
Aber leider bekomm ich die Fehlermeldung unten nicht weg. Hat jemand eine Idee?

 ```
/Users/myuser/Documents/Arduino/libraries/MCCI_LoRaWAN_LMIC_library/src/hal/getpinmap_thisboard.cpp: In function 'const Arduino_LMIC::HalPinmap_t* Arduino_LMIC::GetPinmap_ThisBoard()':
/Users/myuser/Documents/Arduino/libraries/MCCI_LoRaWAN_LMIC_library/src/hal/getpinmap_thisboard.cpp:71:72: note: #pragma message: Board not supported -- use an explicit pinmap
         #pragma message("Board not supported -- use an explicit pinmap")
                                                                        ^
/Users/myuser/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/myuser/Library/Arduino15/packages/esp32/hardware/esp32/2.0.3-RC1/tools/sdk/esp32/lib/libpp.a(hal_mac.o): in function `hal_init':
(.text.hal_init+0xb4): multiple definition of `hal_init'; libraries/MCCI_LoRaWAN_LMIC_library/hal/objs.a(hal.cpp.o):/Users/myuser/Documents/Arduino/libraries/MCCI_LoRaWAN_LMIC_library/src/hal/hal.cpp:416: first defined here
collect2: error: ld returned 1 exit status
Mehrere Bibliotheken wurden für "WiFi.h" gefunden
 Benutzt: /Users/myuser/Library/Arduino15/packages/esp32/hardware/esp32/2.0.3-RC1/libraries/WiFi
 Nicht benutzt: /Applications/Arduino.app/Contents/Java/libraries/WiFi
exit status 1
Fehler beim Kompilieren für das Board Heltec Wireless Stick.
```

schon mal vielen Dank!

ThomasWaldmann, 2022-04-14:
I didn't test the 2.0.3-rc1 arduino yet. It worked for me with the previous stable release.

It sounds like being caused by duplicated libraries maybe, so some cleanup there (== only having one) could also help.

The "explicit pinmap" warning maybe can get ignored.

zwiebelxxl, 2022-04-24:
after correctly installing python, the code could be compiled.

closed 2022-04-24 by zwiebelxxl