ecocurious2/MultiGeiger #455: Hardware Probleme

zwiebelxxl, 2022-04-24:
Ich habe jetzt alle Bauteile bekommen und verlötet.
Leider startet der esp32 nicht wenn er mit der Platine verbunden ist.

Die Bauteile hab ich nach Stückliste bestellt bis auf den D6 (Gleichrichterdiode GP10Y).
Den konnte ich nirgends bekommen und hab mich für einen anderen entschieden [EM 513](https://www.reichelt.de/gleichrichterdiode-1600-v-1-a-do-5-em-513-p7364.html)

Wenn ich die 3,3 Volt Spannung messe dann sind es von 4,8 bis 4,9 Volt! 
Auf der Hochvolt Seite (Zählrohr) konnte ich bis zu 200 Volt Messen.

Könnte die falsche Diode das Problem sein? Was für alternativen gibt es zu dem Bauteil?

Vielen Dank!

zwiebelxxl, 2022-06-15:
Ich hab mir bei JLCPCB 10 Platinen machen lassen und zwei davon jetzt bestückt, der Fehler ist bei beiden gleich.
Die Lötstellen hab ich mehrfach angeschaut und die Platine zusätzlich gereinigt.

Aber Leider fehlt mir noch das Zählrohr ( hoffe das kommt dieses Jahr noch... ), meine Vermutung ist das es daran liegt....kann das sein?

ituri, 2022-06-15:
Hier das gleiche: Ich habe in meiner Verzweiflung eine zweite Platine nochmal mit neuen Komponenten bestückt - gleiches Verhalten. 
Lötstellen alle mehrfach gecheckt.

rexfue, 2022-06-23:
Also wenn Ihr wollt, könnt Ihr mir eine der Platinen mal zusenden, dann werde ich das mal untersuchen. Schreibt mit ne PM, dann sende ich die Adresse.

rexfue, 2022-06-25:
@ituri : Hast Du die Platinen selber machen lassen oder welche von uns (von Jürgen). Im ersten Fall hast Du wahrscheinlich ein älteres Layout verwendet, das mit den neuen Heltec-Boards noch nicht klar kommt. Heltec hat das Pinning der Boards geändert. Ich habe leider erst heute bemerkt, dass auf GitHub noch alte Dateien liegen (jetzt sind sie aktuell). Beim alten Layout müssen zwei Leitungen gekappt werden - siehe beiliegendes Foto.[image]

ituri, 2022-06-25:
@rexfue Danke, das ist schon mal ein guter Hinweis. Ich hatte in der Tat die Platine von GitHub genommen und bei jlpcb drucken lassen. Es war die alte Version. Habe nun mit einem Skalpell die Leiterbahnen durchgetrennt und auch per Durchgangsprüfung mit dem Multimeter überprüft, ob noch eine Verbindung besteht. Tut sie nicht. Zum Vergleich habe ich sogar ein unbestücktes PCB verwendet, dort kann ich auf den beiden gezeigten Bahnen einen Durchgang messen.

Trotzdem bleibt das Problem: Sobald ich den ESP einsetze, bleibt das Display schwarz. Nehme ich ihn heraus, startet die Firmware und zeigt Phantasiewerte an.

Gibt es womöglich noch andere Änderungen am PCB, die ich umsetzen muss?

rexfue, 2022-06-25:
Hmm, ich habe das gerade eben bei zwibelxxls Paltinen so durchgeführt und es hat geklappt. Andere Änderungen gibt es im Layout nicht. Miss mal die Spannungen - insbesondere 5V und 3.3V nach. Ansonsten kann ich nur nochmal anbieten: Platine(n) her schicken, ich kann mal nachschauen.

ituri, 2022-06-25:
Auf das Angebot komme ich gerne zurück. Eine Rückfrage allerdings vorher noch: Wenn ich den ESP aus der Stiftleiste nehme und mit dem Multimeter messe, sind an 3.3V eben jene 3.3V und an 5V die 5V. Wenn er in der Stiftleiste steckt, messe ich jedoch auch beim 3.3V-Pin die gleiche Spannung wie beim 5V-Pin, nämlich ca. 5.1V. Das sollte nicht sein, oder?

rexfue, 2022-06-25:
Nee, das sollte nicht sein

ituri, 2022-06-25:
Ok, Du hast Post von mir. 😊