ecocurious2/MultiGeiger #456: Wifi issue

zwiebelxxl, 2022-04-24:
the esp32 is not connecting to wlan. 

I have a 36 digit long wlan password with special signs like "$" and "_". 
Is there a limitation on it?

If i open a guest wlan it connections with no issues. There is a shorter and easier password configured.

Logs:

11:05:11.431 -> GEIGER: 1970-01-01T00:00:00 ------------------------------------------------------------------------------------------------------------------------
11:05:11.431 -> GEIGER: 1970-01-01T00:00:00 Simple Multi-Geiger, Version V1.16.0
11:05:11.431 -> GEIGER: 1970-01-01T00:00:00 ------------------------------------------------------------------------------------------------------------------------
11:05:11.469 -> State changing from: 0 to 2
11:05:11.609 -> Setting up AP: ESP32-16687980
11:05:11.609 -> Use password: <hidden>
11:05:11.609 -> AP IP address: 192.168.4.1
11:05:11.609 -> AP timeout (ms): 30000
11:05:11.609 -> State changed from: 0 to 2
11:05:41.606 -> State changing from: 2 to 3
11:05:41.716 -> Connecting to [4b] (password is hidden)
11:05:41.716 -> WiFi timeout (ms): 30000
11:05:41.716 -> State changed from: 2 to 3
11:06:11.724 -> Giving up.
11:06:11.724 -> State changing from: 3 to 2
11:06:11.724 -> Setting up AP: ESP32-16687980
11:06:11.724 -> Use password: <hidden>
11:06:11.724 -> AP IP address: 192.168.4.1
11:06:11.724 -> AP timeout (ms): 30000

rexfue, 2022-04-24:
yes. We are using third party software IotWebConfig and they limit standard password to 32 characters.

zwiebelxxl, 2022-04-24:
Thanks! Now it's working! Many thanks for your response!

closed 2022-04-24 by zwiebelxxl