ecocurious2/MultiGeiger #461: Software  Error: Board not supported / multiple definition of `hal_init'

norbi919, 2022-07-04:
Hallo,

bekomme beim Kompilieren erst eine Warnung:

hal.cpp:
C:\Users\Norbert\Documents\Arduino\libraries\arduino-lmic-master\src\hal\getpinmap_thisboard.cpp: In function 'const Arduino_LMIC::HalPinmap_t* Arduino_LMIC::GetPinmap_ThisBoard()':
C:\Users\Norbert\Documents\Arduino\libraries\arduino-lmic-master\src\hal\getpinmap_thisboard.cpp:71:72: note: #pragma message: Board not supported -- use an explicit pinmap
         #pragma message("Board not supported -- use an explicit pinmap")

später dann ein Exit:

kurz vorm Ende des Compilevorgangs:
c:/users/norbert/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\Norbert\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3-RC1/tools/sdk/esp32/lib\libpp.a(hal_mac.o): in function `hal_init':
(.text.hal_init+0xb4): multiple definition of `hal_init'; C:\Users\Norbert\AppData\Local\Temp\arduino_build_964013\libraries\arduino-lmic-master\hal\objs.a(hal.cpp.o):C:\Users\Norbert\Documents\Arduino\libraries\arduino-lmic-master\src\hal/hal.cpp:416: first defined here
collect2.exe: error: ld returned 1 exit status

Kompiliere mit der Arduinoapp und den aktuellsten Libraries.
Die MCCI LoRaWAN library and die ESP32 base libraries haben da irgendwie ein Problem.

Gibt es evtl. Libversionen die entsprechend zusammen passen?

Grüße, Norbert.

rexfue, 2022-07-07:
Das liegt wohl an der neuesten Espressif32-Arduino-Anpassung. Es gibt zwei Workarounds:
- entweder auf eine ältere Version der ESP32-Libraries zurückgehen (bei Platformio z.b. in platformio.ini einfügen: platform = espressif32@3.5.0
- oder aber den LMIC Library Ordner in das lib/ Verzeichnis des Projektes kopieren und da dann hal_init() überall, wo es vorkommt (ist nur 2x), umbenennen. Ist nicht elegant, aber geht erst mal.

ThomasWaldmann, 2023-10-15:
1 Jahr später... - ist das Problem immer noch nicht richtig gelöst?

rexfue, 2023-10-16:
It IS solved, use 

build_flags =
  -D CFG_eu868=1
  -D CFG_sx1276_radio=1
  -D ARDUINO_LMIC_PROJECT_CONFIG_H_SUPPRESS=1
  -D hal_init=LMICHAL_init                                <---------------------

in platform.ini.

ThomasWaldmann, 2023-10-16:
I have that (have copied git-committed `platformio-example.ini` to local `platformio.ini`), but as I use the arduino IDE, I guess that is not used?

rexfue, 2023-10-16:
You are right. If you find, how to set precompiler flags in Arduino IDE, then you can try it

ThomasWaldmann, 2023-10-16:
Didn't find a way yet. There is no GUI dialogue.

Also tried this script to invoke arduino ide, but the compilation still does not work:

```
export CFLAGS="-DCFG_eu868=1 -DCFG_sx1276_radio=1 -DARDUINO_LMIC_PROJECT_CONFIG_H_SUPPRESS=1 -Dhal_init=LMICHAL_init"

/Applications/Arduino\ IDE.app/Contents/MacOS/Arduino\ IDE
```

rexfue, 2023-10-16:
Hmmm, ich verwende auf dem MAC Visual Studio Code und Platformio für alle ESP32 Anwendungen