ecocurious2/MultiGeiger #464: support of platformio: E (554) spi_flash: Detected size(4096k) smaller than the size in the binary image header(8192k).

rwe17, 2022-09-12:
I'm trying to build and upload the Multigeige v1.16.0 firmware into a Heltec wifi_kit_32 device using the Platformio IDE.

Unfortunately, from the documentation i only found a way using the Arduion IDE which does not work on my Linux machine, because of several compilation issues event though i imported all required libraries and applied the patch file.

For me it seem more easy and logical to use Platformio and a `platformio.ini` file that describes all prerequisites upfront. After solving the issue with the redundant `hal_init()` function, i was able to upload the firmware into my esp32 board. Unfortunately, the defined [board configuration](https://docs.platformio.org/en/latest/boards/espressif32/heltec_wireless_stick.html) in platformio.ini `board = heltec_wireless_stick` obviously does not fit to the hardware, because this board defines a 8 MB Flash size and my board has 4MB which results in endless reboots showing this error:

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
E (554) spi_flash: Detected size(4096k) smaller than the size in the binary image header(8192k). Probe failed.

assert failed: do_core_init startup.c:298 (flash_ret == ESP_OK)
```

Based on this results i have three questions:
a) Is it possible to configure the configuration platformio.ini to coerce the flashing tool to create 4MB fash size packages for the board heltec_wireless_stick? (applying `board_build.partitions = min_spiffs.csv` as described in #449 does not solve the issue.)
b) When changing the platformio.ini to `board = heltec_wifi_kit_32` i run into several compilation issues. Is it possible to fix that in the source code? I guess when selecting the proper board, the firmware will run and start without infinite reboots.
c) In case Platformio is not an option because its to difficult for the existing code, is it possible to upload the binary release file just via a tool from the command line? Which tool is required and how should i call that?

Any support is appreciated to get the stuff running on my heltec wifi_kit_32.

Regard
Ralf

rexfue, 2022-09-13:
Please send me (rxf@gmx.de) an email. I will try to answer your questions by mail. After solving the issues i will post the results here.

rwe17, 2022-09-18:
Ok, my board is now running - thanks to the support of rexfue!

Let's tell the long story short:

__4MB Flash size:__
* I bought an "old" heltec `wifi_kit_32` that supports only 4MB flash size. 
* The platformio-example.ini specifies the `heltec_wireless_stick` that has configured a 8MB flash size
* To solve that, you need to specify the board `heltec_wifi_lora_32` (which reflects my old heltec device) and you need to add the `board_build.partitions = min_spiffs.csv`

__Endless Rebooting:__
After this was fixed and the firmware could be uploaded into my old heltec esp32, the device restarted in an infinit loop.
* The reason for that behavior was a too new _espressif32_ platform version i've used as default in my Platformio. This too new platform version caused a `Guru Meditation Error: Core  1 panic'ed (LoadProhibited)` when setting up the interrupt routines for the tube.
* To fix that you just need to specify explicitly the version 3.5.0: `platform = espressif32@3.5.0`

So at the end, my platformio.ini [env:geiger] section looks like that:

```
[env:geiger]
# select 'heltec_wifi_lora_32' to support the "old" 4Mb Heltec wifi_kit_32 
board = heltec_wifi_lora_32
board_build.partitions = min_spiffs.csv
build_flags =
  -D CFG_eu868=1
  -D CFG_sx1276_radio=1
  -D ARDUINO_LMIC_PROJECT_CONFIG_H_SUPPRESS=1
  -D hal_init=LMICHAL_init # handles multiple hal_init()
# stick to 3.5.0, because newer versions cause an issue when setting up the interrupt routines
platform = espressif32@3.5.0
framework = arduino
monitor_speed=115200
lib_deps=
  U8g2
  Adafruit BME680 Library@^2.0.0
  Adafruit BME280 Library
  Adafruit Unified Sensor
  IotWebConf@^3.1.0
  MCCI LoRaWAN LMIC library
  h2zero/NimBLE-Arduino
```