ecocurious2/MultiGeiger #465: hardware source ?

hannesa2, 2022-09-15:
Sicher schon 100x gefragt: Ich will mir das Löten sparen und mir die fertige hardware kaufen.
Gibt es so was und wenn ja, wo bekomme ich die her ?

zwiebelxxl, 2022-10-25:
Hallo,
ich hab mir 10 Platinen in China machen lassen und die Bauteile aus verschiedenen quellen zusammen gekauft.
Bis auf den esp32 und das Zählrohr könnte ich dir helfen....

hannesa2, 2022-10-25:
Danke, aber wenn mit die beiden Dinger abgehen, dann bin ich verloren, da ich dann wieder eine Extrawurst haben werde und wieder irgendwo abweiche. Fehlersuche ist dann blöd, wenn man keine funktionierende Referenz hat. 
Das ist genau das, was ich gerne hätte: eine funktionierende Referenz !
Das ist zumindest meine Angst.