ecocurious2/MultiGeiger #467: Clicking and blinking stops after some time.

pbriesch, 2023-05-13:
Hello,

the clicking of the speaker and the blinking of the LED stops after some time. A reboot or going to the config_page and clicking on **Apply** makes ist work again. The debug output shows that still events are being counted and uploaded.

Furthermore, after clicking **Apply** I come to a page which shows `Configuration saved. Return to [home page](http://192.168.4.194/).` When I the go to the home page, the clicking and flashing of the LED stops.  Events are still being counted and uploaded.

I'm using the master branch compiled with platformIO and the Heltec wireless stick with LoRa.

Regards

ThomasWaldmann, 2023-05-13:
Don't know why clicking stops for you. Can you be more specific than "after some time"?

But the config page / saving the config is special:
it disables the clicking/blinking as there could be some interference with the flashing process (IIRC) and after finishing, it enables the clicking/blinking again.

pbriesch, 2023-05-13:
It usually happens after about 2 to 3 hours. Not after half a day or days.

pbriesch, 2023-05-16:
@ThomasWaldmann 

I found out that the clicking and blinking doesn't stop after some time if **only** both `Send to madavi.de` and `Send to LoRa (=>TTN)` are selected.
As soon as I select `Send to BLE (Reboot required!)` in addition to the prior selections it will stop after some time.
I didn't select `Send to sensor.community` because I transfer the data by LoRaWAN.

Maybe this helps to find the cause of this behaviour?