ecocurious2/MultiGeiger #468: DIP-switches don't work

pbriesch, 2023-05-13:
In order to use the DIP-switches I cut the fine bridges on the PCB. Do I need to add the 10k pull-ups as well to make it work?

Independent from the settings, the debug output is always this:

`GEIGER: 1970-01-01T00:00:00 Switches: SW0 speaker_on 1,  SW1 display_on 1,  SW2 led_on: 1,  SW3 ble_on: 1`

From switches.cpp:
```
  pinMode(PIN_SWI_0, INPUT);  // These pins DON'T HAVE PULLUPS!
  pinMode(PIN_SWI_1, INPUT);
  pinMode(PIN_SWI_2, INPUT);
  pinMode(PIN_SWI_3, INPUT);

  Switches s = read_switches();
  log(DEBUG, "Switches: SW0 speaker_on %d,  SW1 display_on %d,  SW2 led_on: %d,  SW3 ble_on: %d",
      s.speaker_on, s.display_on, s.led_on, s.ble_on);
}

Switches read_switches() {
  Switches s;

  // Read Switches (active LOW!)
  s.speaker_on = !digitalRead(PIN_SWI_0);
  s.display_on = !digitalRead(PIN_SWI_1);
  s.led_on = !digitalRead(PIN_SWI_2);
  s.ble_on = !digitalRead(PIN_SWI_3);
```

pbriesch, 2023-05-13:
While experimenting with the code a bit, I found that if I explicitly turn off the LED with the following code

`s.led_on = 0;`

The LED no longer flashes, but lights up constantly. I expected the white LED to be dark.

pbriesch, 2023-05-13:
When I explicitly turn off the display with 
`s.display_on = 0;`
after reboot the display turns on for a second and then turns off as expected. However, after 1 or 2 seconds, it turns on again and stays on.

pbriesch, 2023-05-17:
The pull-up resistors are obviously need, see [https://github.com/ecocurious2/MultiGeiger/pull/369#issuecomment-760211034](https://github.com/ecocurious2/MultiGeiger/pull/369#issuecomment-760211034)

t-pi, 2023-08-06:
Actually, IIRC, the switches are just there for inital configuration upon first boot, but are overwritten as soon as a config is stored and re-read after follow-up boots. Otherwise, a web config interface would not make much sense.
That's why I stopped populating my Multigeigers' DIP-switches after the first few...

But yes, the pull-ups are needed, if you want to use them.

pbriesch, 2023-08-06:
Thank you very much for your answer!

closed 2023-08-06 by pbriesch