ecocurious2/MultiGeiger #469: make a new release based on current libraries / firmware

ThomasWaldmann, 2023-10-15:
Got an email from a user with strange wifi problems which suggested that they might be fixed on newer ESP IDF.

So, guess it would be good to make a new release based on the current stuff.

rexfue, 2023-10-15:
I guess, You should do it

ThomasWaldmann, 2023-10-15:
@rexfue yeah, just need to install the current stuff on the macbook first.

MMBytes, 2024-02-17:
Hi @ThomasWaldmann, any chance for a new release? Thanks

ThomasWaldmann, 2024-02-17:
@MMBytes last time I tried I ran into strange build issues I couldn't resolve. 

I'll try again soon....

MMBytes, 2024-02-17:
> @MMBytes last time I tried I ran into strange build issues I couldn't resolve.
> 
> I'll try again soon....

Thank you!

MMBytes, 2024-05-26:
Hi @ThomasWaldmann, any luck with this? 
Many thanks

eku, 2025-03-25:
As I understand it, no version for Espressif's SDK is specified in platformio.ini and therefore the latest version is automatically used. However, this is not compatible with the source code version.

There is nothing wrong with using an older version. But which one?