ecocurious2/MultiGeiger #473: Is Heltec WiFi Kit 32 (V3) supported?

Nestade, 2023-12-25:
Hello,

I just attempted to set up my current MultiGeiger with WiFi to integrate it into the measuring network. Unfortunately, it appears that the WiFi on my Heltec WiFi Kit 32 is not functioning properly, so I may need to get a new one. Does anyone know if I can use the Heltec WiFi Kit 32 (V3), or does it have to be the previous revision?

Regards,
Nestade

rexfue, 2023-12-25:
No, sorry, V3 ist still not supported. You can get a V2-Board from Jürgen ( juergen@boehri.de ). Regards Reinhard

closed 2023-12-25 by Nestade

Nestade, 2023-12-26:
Thanks for the quick reply. I unexpectedly found the issue when I touched the PCB antenna, and it suddenly connected. I'll solder an external antenna to see if that resolves the issue.